jcane86/motor-hat #18: Error when initializing motor-hat for stepper motors

richiepreece, 2017-07-10:
Hi, I'm getting the following error when attempting to initialize my application:

```
/home/pi/intervalometer/node_modules/i2c-bus/i2c-bus.js:257
  i2c.writeByteSync(peripheralSync(this, addr), cmd, byte);
      ^

Error: , Remote I/O error
    at Error (native)
    at Bus.writeByteSync (/home/pi/intervalometer/node_modules/i2c-bus/i2c-bus.js:257:7)
    at setAllPWM (/home/pi/intervalometer/node_modules/motor-hat/dist/pwm.js:148:9)
    at init (/home/pi/intervalometer/node_modules/motor-hat/dist/pwm.js:173:5)
    at exports (/home/pi/intervalometer/node_modules/motor-hat/dist/pwm.js:185:3)
    at init (/home/pi/intervalometer/node_modules/motor-hat/dist/index.js:124:11)
    at exports (/home/pi/intervalometer/node_modules/motor-hat/dist/index.js:142:3)
    at Object.<anonymous> (/home/pi/intervalometer/server/app.js:47:53)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)
```

Any ideas how to go about fixing this? Thanks in advance!

richiepreece, 2017-07-10:
Now, I'm trying to do some development to prepare for a motor hat arriving on Tuesday. Could a lack of motor hat be the issue? Since there's technically no i2c device installed?

jcane86, 2017-07-21:
Hey there,

sorry for the delay, I recently moved house and just got all my setup back in place to test with the rpi. Did you manage to get it working with the actual MotorHAT?
It does sound like it could be a problem with the board not beeing there, but I haven't been able to reproduce it yet.

richiepreece, 2017-07-21:
I wasn't able to get it working. When using Adafruit's example python code, I could get the hat to run, but when using this library, I still had the error thrown. The breaking code looks like:

```
var motorHat = require('motor-hat')({
  steppers: [
    ['M1', 'M2'],
    ['M3', 'M4']
  ]
});
```

jcane86, 2017-07-21:
Hiya,

I was just going through the config, as I don't actually have an official adafruit MotorHAT to test this on, but it seems my (clone) board has a different i2c address than the default adafruit one.

Could you change that line for:

```
var motorHat = require('motor-hat')({
  address: 0x60,
  steppers: [
    ['M1', 'M2'],
    ['M3', 'M4']
  ]
});
```

and see if it makes any difference?

Alternatively, check in Adafruit's python code, in the file Adafruit_MotorHAT_Motors_Py, around line 227 you should have a line similar to this:

```
class Adafruit_MotorHAT:
    [...]
    def __init__(self, addr = 0x60, freq = 1600, i2c=None, i2c_bus=None):

```
If that address (_**addr**_) is not 0x6F (default in my library) change it to whatever's in there (0x60 in this case).

Hope that helps.

Cheers

richiepreece, 2017-07-21:
Ok. I'll give it a go. Seeing address `0x60` seems to ring a bell, so I'll let you know if that helps. Thanks again!

jcane86, 2017-08-09:
Hi @richiepreece , 

any luck with this? should I close the issue?

richiepreece, 2017-08-09:
Hey, my brother in law has the hat, so I've been unable to test. Sorry! Feel free to close the issue

liabru, 2017-12-25:
Setting the address solved this issue for me. Thank you for your work on this!

GitStevie, 2018-02-05:
Just wanted to say I had the exact same problem above.  I came here to whine about it and saw the post above with the identical issue. 
 Based on that I tried the "address: 0x60", which I hadn't seen in the documentation and poof - it worked!  Coolio!  Thanks! I have two Adafruit stepper hats and default i2c addresses it uses are 0x60 and 0x70.  Thanks again!

closed 2018-02-27 by richiepreece

violuke, 2018-09-13:
@jcane86 Thanks for the handy lib. I can see that this is referenced in the readme, but it would have been really handy for me to have known that the "non-default" was the "default" for the official hat. Could that be added to the readme for others? I wasted a lot of time on this step. Thanks.

jcane86, 2018-09-13:
@violuke Yeah, point taken. I started this for myself, and I don't have an official hat, so it kinda grew from there. 

Do you think there's a better way to default this? I'm a bit hesitant to just make the official one default, as it would break it for anyone using the default. Maybe that's not a lot of people anyway.

Otherwise, feel free to open a PR with some more clarification to the readme, that would be much appreciated.

Thanks for your feedback!

commit fd29d58c (2018-02-28): docs(Readme): add reference to i2c address 

Update README.md
Fixes #18