jcane86/motor-hat #25: TypeError: Cannot read property 'PWM' of undefined

Triadager, 2017-08-31:
Hello, 

I installed the module and tried to run the example code that was provided. But I keep getting this error: 

`pi@raspi:~/RaspiStepperTests $ node StepperHatTest.js
/home/pi/RaspiStepperTests/node_modules/motor-hat/dist/index.js:90
      pinAloc[pins[stepper.W1].PWM] = true;
                              ^

TypeError: Cannot read property 'PWM' of undefined
    at /home/pi/RaspiStepperTests/node_modules/motor-hat/dist/index.js:90:31
    at Array.map (<anonymous>)
    at checkPins (/home/pi/RaspiStepperTests/node_modules/motor-hat/dist/index.js:89:17)
    at init (/home/pi/RaspiStepperTests/node_modules/motor-hat/dist/index.js:127:5)
    at exports (/home/pi/RaspiStepperTests/node_modules/motor-hat/dist/index.js:143:3)
    at Object.<anonymous> (/home/pi/RaspiStepperTests/StepperHatTest.js:12:36)
    at Module._compile (module.js:573:30)
    at Object.Module._extensions..js (module.js:584:10)
    at Module.load (module.js:507:32)
    at tryModuleLoad (module.js:470:12)
`

What am I doing wrong? It seems it cannot even initiate the motor.

Moving the servo works (even though I think not over its whole angular range, but at least it moves).

I am really new to robotics and even the Pi itself, and hours and hours of googling did not get me anywhere, so I'm sorry to bother. 

Kind regards, 

Triadager

jcane86, 2017-08-31:
Hi Triadager, 

Could you show me the code you're running? Is it the exact one from the Readme? 

Cheers

Triadager, 2017-08-31:
Yeah, it basically is. I just removed the dc motor and one of the servos. But I also just ran both the copypasted code and only the initialization of the stepper, both gave me the same error. I'm pretty clueless. 

Thanks for the quick reply by the way, you're awesome!

Cheers, 

Triadager

Triadager, 2017-08-31:
Is there anything I need besides a current version of node.js, the module, and i2c enabled?

jcane86, 2017-08-31:
OK, I got it,

seems my README sample code is slightly wrong. It must have been left from a previous version.

I'll correct the sample or make the code work with it, but in the meantime, if you change the line:

`        steppers: [['M1', 'M2']],`

to 

`        steppers: [{ W1: 'M1', W2: 'M2' }],`

it should work.

By the way, could you please tell me if you're running on an "official" Adafruit MotorHat or another brand? I suspect my default i2c address doesn't work on "official" boards, but I don't have one to test it.

Cheers

jcane86

closed 2017-09-01 by jcane86

Triadager, 2017-09-01:
Nice, it worked! Thanks, I really appreciate your help. 
About the board: To be honest, I use a knock-off from AliExpress, my student budget is quite tight and original Adafruit stuff is kinda expensive over here in Germany. =/

Thanks again though! Have a nice day.

Cheers,

Triadager

jcane86, 2017-09-01:
Cool, glad to help.

Same thing with the original adafruit boards here in UK, lol, that's why I'm looking for someone who has one to test it for me =).

About the servos, take a look at your servos dataset if you have one, and play around with the bit of sample code for 'servo calibration' to set the end positions. You should be able to get the whole range of motion.

Word of warning, there's only one pwm chip in the board, and the steppers and servos use different pwm frequencies, so using one type of motor might fuck up the configuration for the other one. It's a bit tricky to get both working at the same time.

Have fun, and don't hesitate to ask if you have difficulties.

Cheers

jcane86

jcane86, 2017-09-01:
*servos datasheet

jcane86, 2017-09-01:
*servos datasheet

commit 79366105 (2017-09-01): docs(README.md): Correct stepper specification sample

Fixes #25