jcane86/motor-hat #34: Hold Mode? - Stepper gets too hot

sgsellmann-miovision, 2017-10-01:
My motor is getting hot while sitting idle (currently 50.7 C). Can a Hold Mode be implemented to reduce or turn-off the holding current when stationary?

I've read https://forums.adafruit.com/viewtopic.php?f=25&t=41048 and https://forums.adafruit.com/viewtopic.php?f=31&t=27322&p=140423 on the issue, which may be of use.

My application will require 24/7 run time and the majority of that no motion is needed and no holding force is required.

Thanks!

jcane86, 2017-10-03:
Hi @sgsellmann-miovision,
I'll definitely look into that, should be a pretty easy add. Just to clarify, I'll add a release() method. As holding is what's actually happening now, when the current keeps being applied to hold the motor in place.

Since we're at it I could add a method to throttle the current down programmatically to control torque and heating. I don't have a measuring setup to debug this at the moment, could use your help if you do.

Cheers

jcane86

sgsellmann-miovision, 2017-10-03:
Yes, that sounds like exactly like what I want! I can measure temp easily and accurately, which would be a good indication of the current.

Thanks!

closed 2017-10-03 by jcane86

sgsellmann-miovision, 2017-10-04:
Excellent work, thanks. I'll try it tonight.

jcane86, 2017-10-04:
Sorry, the release function is actually called releaseSync().

Im working to get all the async methods into master

sgsellmann-miovision, 2017-10-04:
Test and working as expected. The temperature now holds steady at ambient when idle.

PR #36 (2017-10-03): feat(stepper.js): Add current control.
Added support to release the motor with release() to stop applying curret to the coils. Or
current(x) to throttle current output by rate x from 0 to 1.

Closes #34