jcane86/motor-hat #56: parambulator error parsing default$ when validating options

PrimeEuler, 2018-03-09:
/home/pi/Documents/Node Projects/node_modules/motor-hat/dist/index.js:42
    throw new Error(err);
    ^

Error: Error: default$
    at errHdlr (/home/pi/Documents/Node Projects/node_modules/motor-hat/dist/index.js:42:11)
    at wrapcb (/home/pi/Documents/Node Projects/node_modules/parambulator/parambulator.js:850:18)
    at /home/pi/Documents/Node Projects/node_modules/parambulator/parambulator.js:968:11
    at Object.fail (/home/pi/Documents/Node Projects/node_modules/parambulator/parambulator.js:592:12)
    at validatedefaults (/home/pi/Documents/Node Projects/node_modules/parambulator/parambulator.js:939:29)
    at Object.Parambulator.self.validate (/home/pi/Documents/Node Projects/node_modules/parambulator/parambulator.js:966:7)
    at MotorHat (/home/pi/Documents/Node Projects/node_modules/motor-hat/dist/index.js:298:11)
    at Object.<anonymous> (/home/pi/Documents/Node Projects/sheldon.js/example/cli/shell.js:9:36)
    at Module._compile (module.js:662:30)
    at Object.Module._extensions..js (module.js:673:10)

PrimeEuler, 2018-03-09:
When running Advanced usage code:

let spec = {
    address: 0x60,
    steppers: [{ W1: 'M1', W2: 'M2' }],
    dcs: ['M4'],
    servos: [0,14]
};
var motorHat = require('motor-hat')(spec);

Address is 0x60

 sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --

jcane86, 2018-03-10:
Hey mate, sorry but I can't reproduce your issue. Do you get it with just those lines of code?

maybe you have an old version of some dependency. do you have a lock file I could look at?

PrimeEuler, 2018-03-10:
Sorry. It looks to be related to my package sheldon.js which also uses
lodash. Your lines of code alone do not produce the error. It only occus if
I load my package before yours, so I beleive its a lodash version issue.
The issue can be closed.

closed 2018-03-10 by PrimeEuler