jcane86/motor-hat #91: busnum is not passed to pwmlib

SovGVD, 2019-04-07:
Hi. Trying to made this library work for nanopi m4 with nanonat motor and found a bug that `busnum` variable is not in opts for pmwlib.
If it is possible, can you please add it into code to avoid extra patches for non-raspberry platforms?

```
diff --git a/lib/index.js b/lib/index.js
index 8187714..591a7cf 100644
--- a/lib/index.js
+++ b/lib/index.js
@@ -211,7 +211,7 @@ module.exports = function MotorHat(opts) {
    * or in second parameter to callback if callback provided, to enable chaining.
    */
   const init = function init(cb) {
-    const pwmopts = { i2c, address: options.address };
+    const pwmopts = { i2c, address: options.address, busnum: options.busnum };
     const self = this;
     pwm = pwmlib(pwmopts);
 
```

Thanks.

SovGVD, 2019-04-21:
Well, looks like does not matter, as there is also different pinout for NanoPO motor HAT.
```
  var pins = {
    M1: {
      PWM: 0,//8,
      IN2: 1,//9,
      IN1: 2,//10
    },
    M2: {
      PWM: 5,//13,
      IN2: 4,//12,
      IN1: 3,//11
    },
    M3: {
      PWM: 15,//2,
      IN2: 14,//3,
      IN1: 13,//4
    },
    M4: {
      PWM: 10,//7,
      IN2: 11,//6,
      IN1: 12,//5
    }
  };
```

And as some of the value is `0`, it cause and issue in https://github.com/jcane86/motor-hat/blob/master/lib/dc.js#L288 as PWM value become `undefined`

jcane86, 2019-04-22:
Hi mate, sorry for the delay in replying, I've been kinda busy.

Thanks for the issue, that's a good catch.

As for the pin definition, I'll try to look into it, but I can't promise I'll be quick.

For now, I would recommend changing the pin definition like so:

motorHat.pins = {
    M1: [0,1,2],
    M2: [5,4,3],
    M3: [15,14,13],
    M4: [10,11,12]
  };

I'm pretty sure that should work for DCs. Don't change the motor definitions if you're going to use steppers or servos, as that might not work, but for DCs it should be fine.

Let me know if you have more trouble with this, and if you feel like making a Pull Request to fix this more permanently, I'd appreciate it.

Cheers

jcane86, 2019-04-22:
:tada: This issue has been resolved in version 2.0.7 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/motor-hat)
- [GitHub release](https://github.com/jcane86/motor-hat/releases/tag/v2.0.7)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #93 (2019-04-22): fix(pwm): pass busnum to pwmlib
closes #91
commit 48eb5b83 (2019-04-22): fix(pwm): pass busnum to pwmlib

closes #91
closed 2019-04-22 by jcane86