jcane86/motor-hat #120: Installation failing on Raspberry Pi 4, node 13.9.0, npm 6.13.7

ksteinberg, 2020-03-04:
> i2c-bus@1.2.5 install /home/pi/Public/agro/node_modules/motor-hat/node_modules/i2c-bus
> node-gyp rebuild

gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/13.9.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/Public/agro/node_modules/motor-hat/node_modules/i2c-bus/.node-gyp"
make: Entering directory '/home/pi/Public/agro/node_modules/motor-hat/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
In file included from ../../nan/nan_converters.h:67,
                 from ../../nan/nan.h:202,
                 from ../src/i2c.cc:2:
../../nan/nan_converters_43_inl.h: In static member function ‘static Nan::imp::ToFactoryBase<v8::Boolean>::return_t Nan::imp::ToFactory<v8::Boolean>::convert(v8::Local<v8::Value>)’:
../../nan/nan_converters_43_inl.h:18:69: error: no matching function for call to ‘v8::Value::ToBoolean(v8::Local<v8::Context>)’
       val->To ## TYPE(v8::Isolate::GetCurrent()->GetCurrentContext())          \
                                                                     ^
../../nan/nan_converters_43_inl.h:22:1: note: in expansion of macro ‘X’
 X(Boolean)
 ^

Log file attached.

Seems to be a problem with the i2c-bus module

pi@raspberrypi:~/Public/agro $ uname -r
4.19.97-v7l+
pi@raspberrypi:~/Public/agro $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~/Public/agro $

[motor1.log](https://github.com/jcane86/motor-hat/files/4287337/motor1.log)

jcane86, 2020-03-04:
hi @ksteinberg ,

Thanks for the report. Unfortunately, I dont have the hardware or the time at the moment to debug issues with the i2c stuff (left the rpi behind in a move, and its the only bit I can't test in CI).

Would you mind installing from this branch instead https://github.com/jcane86/motor-hat/tree/greenkeeper/i2c-bus-2.0.0 and giving it a try? contrary to the name, it upgrades i2c-bus to 5.1.0. Been wanting to merge that, but I dont currently have any way to test it.

Thanks, that'd be a great help to finally close #50

ksteinberg, 2020-03-05:
First a disclaimer...not an expert on installing libraries directly via git but here is what I did.

**** would not run because it claimed it could not find the bindings.

1) Copied the git branch down to my system as a Zip
2) Unzipped and copied it to a directory on my Pi (via Filezilla)
3) Ran "sudo npm install ../motor-hat-greenkeeper-i2c-bus-2.0.0/"

**Output
npm WARN agro@1.0.0 No description
npm WARN agro@1.0.0 No repository field.

+ motor-hat@1.0.0-semanticrelease
added 1 package from 1 contributor and audited 677 packages in 3.616s
found 6 vulnerabilities (2 low, 4 high)
  run `npm audit fix` to fix them, or `npm audit` for details

4) ran "sudo node pod-light-test.js"

** Output
pi@raspberrypi:~/Public/agro $ sudo node pod-light-test.js
1583425528096 Available Raspi IO
/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/Debug/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/Release/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/out/Debug/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/Debug/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/out/Release/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/Release/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/default/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/compiled/13.9.0/linux/arm/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/addon-build/release/install-root/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/addon-build/default/install-root/i2c.node
 → /home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/lib/binding/node-v79-linux-arm/i2c.node
    at bindings (/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/bindings/bindings.js:126:9)
    at Object.<anonymous> (/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/i2c-bus.js:4:32)
    at Module._compile (internal/modules/cjs/loader.js:1151:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1171:10)
    at Module.load (internal/modules/cjs/loader.js:1000:32)
    at Function.Module._load (internal/modules/cjs/loader.js:899:14)
    at Module.require (internal/modules/cjs/loader.js:1040:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/lib/index.js:13:13)
    at Module._compile (internal/modules/cjs/loader.js:1151:30) {
  tries: [
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/Debug/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/Release/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/out/Debug/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/Debug/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/out/Release/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/Release/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/build/default/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/compiled/13.9.0/linux/arm/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/addon-build/release/install-root/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/addon-build/debug/install-root/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/addon-build/default/install-root/i2c.node',
    '/home/pi/Public/motor-hat-greenkeeper-i2c-bus-2.0.0/node_modules/i2c-bus/lib/binding/node-v79-linux-arm/i2c.node'
  ]
}

More Info
******************
npm list 
[... 324 lines cut ...]
-rw-r--r--   1 pi pi  6698 Feb 11 11:05 piserver.js
-rw-r--r--   1 pi pi  4275 Mar  5 11:08 pod-light-test.js
-rw-r--r--   1 pi pi   467 Mar  1 15:37 relay-test.py
-rw-r--r--   1 pi pi   132 Feb 11 09:48 servo.py
-rw-r--r--   1 pi pi   423 Mar  3 15:20 stemma.py

***********
System Info
*************

pi@raspberrypi:~/Public/agro $ sudo node -v
v13.9.0
pi@raspberrypi:~/Public/agro $ sudo npm -v
6.14.2
pi@raspberrypi:~/Public/agro $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

jcane86, 2020-03-05:
hi @ksteinberg,

for reference, you can add packages directly from github like this: npm install github:jcane86/motor-hat#greenkeeper-i2c-bus-2.0.0

as for the error, it looks like something is wrong with the build... maybe having it installed like that caused some problem. please try the following:
- uninstall your local copy of motor-hat: npm uninstall motor-hat
- delete your node_modules folder (to start from a clean state) : rm -rf node_modules
- install motor_hat from github: npm install github:jcane86/motor-hat#greenkeeper-i2c-bus-2.0.0
- try again 🤞

ksteinberg, 2020-03-06:
Will do...thanks for help/

I was trying to install from github but was having a issue.  I will retry...

ksteinberg, 2020-03-06:
still failing... :(

I created a whole new clear project...no other modules except motor-hat.  Downloaded directly from GIT.  

I hate to ask but if your buried and can't get to this, do you recommend another npm for this?  Not sure how else to control the hat.

Happy to keep doing QA if you want to make tweaks.

----------------------------------
Results
-----------------------------------

pi@raspberrypi:~/Public/servo $ sudo npm install github:jcane86/motor-hat#greenkeeper/i2c-bus-2.0.0

> i2c-bus@5.1.0 install /home/pi/Public/servo/node_modules/i2c-bus
> node-gyp rebuild

gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/13.9.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/Public/servo/node_modules/i2c-bus/.node-gyp"
make: Entering directory '/home/pi/Public/servo/node_modules/i2c-bus/build'
  CXX(target) Release/obj.target/i2c/src/i2c.o
  SOLINK_MODULE(target) Release/obj.target/i2c.node
  COPY Release/i2c.node
make: Leaving directory '/home/pi/Public/servo/node_modules/i2c-bus/build'

> sleep@6.1.0 install /home/pi/Public/servo/node_modules/sleep
> node-gyp rebuild

gyp WARN EACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/13.9.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/Public/servo/node_modules/sleep/.node-gyp"
make: Entering directory '/home/pi/Public/servo/node_modules/sleep/build'
  CXX(target) Release/obj.target/node_sleep/module_init.o
In file included from ../sleep.h:4,
                 from ../module_init.cc:1:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
In file included from ../../nan/nan.h:54,
                 from ../sleep.h:4,
                 from ../module_init.cc:1:
../module_init.cc: At global scope:
/home/pi/Public/servo/node_modules/sleep/.node-gyp/13.9.0/include/node/node.h:618:43: warning: cast between incompatible function types from ‘void (*)(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’ {aka ‘void (*)(v8::Local<v8::Object>)’} to ‘node::addon_register_func’ {aka ‘void (*)(v8::Local<v8::Object>, v8::Local<v8::Value>, void*)’} [-Wcast-function-type]
       (node::addon_register_func) (regfunc),                          \
                                           ^
/home/pi/Public/servo/node_modules/sleep/.node-gyp/13.9.0/include/node/node.h:652:3: note: in expansion of macro ‘NODE_MODULE_X’
   NODE_MODULE_X(modname, regfunc, NULL, 0)  // NOLINT (readability/null_usage)
   ^~~~~~~~~~~~~
../module_init.cc:24:1: note: in expansion of macro ‘NODE_MODULE’
 NODE_MODULE(node_sleep, init)
 ^~~~~~~~~~~
  CXX(target) Release/obj.target/node_sleep/sleep_cpp11.o
In file included from ../sleep.h:4,
                 from ../sleep_cpp11.cc:1:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
  CXX(target) Release/obj.target/node_sleep/sleep_posix.o
In file included from ../sleep.h:4,
                 from ../sleep_posix.cc:1:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
  CXX(target) Release/obj.target/node_sleep/sleep_win.o
In file included from ../sleep.h:4,
                 from ../sleep_win.cc:1:
../../nan/nan.h: In function ‘void Nan::AsyncQueueWorker(Nan::AsyncWorker*)’:
../../nan/nan.h:2298:62: warning: cast between incompatible function types from ‘void (*)(uv_work_t*)’ {aka ‘void (*)(uv_work_s*)’} to ‘uv_after_work_cb’ {aka ‘void (*)(uv_work_s*, int)’} [-Wcast-function-type]
     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete)
                                                              ^
  SOLINK_MODULE(target) Release/obj.target/node_sleep.node
  COPY Release/node_sleep.node
make: Leaving directory '/home/pi/Public/servo/node_modules/sleep/build'
npm WARN saveError ENOENT: no such file or directory, open '/home/pi/Public/servo/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/home/pi/Public/servo/package.json'
npm WARN servo No description
npm WARN servo No repository field.
npm WARN servo No README data
npm WARN servo No license field.

+ motor-hat@1.0.0-semanticrelease
added 15 packages from 21 contributors and audited 16 packages in 26.449s
found 6 vulnerabilities (2 low, 4 high)
  run `npm audit fix` to fix them, or `npm audit` for details

jcane86, 2020-03-07:
hi @ksteinberg , thanks for testing that. OK, at least it's not failing in i2c-bus now.

I have a few ways in mind I could just remove the dependency on sleep altogether, I'm gonna have to work on this a bit. I should have something working in a couple days. Will let you know

jcane86, 2020-03-07:
actually, I'm just thinking, node LTS is currently node 12, and 13 is one of the unstable versions that won't even be LTS, so potentially doing a major revision for node 13 seems unnecessary.

@ksteinberg would you be able to do these tests with node 12? that's the latest version I'm currently testing the lib on

ksteinberg, 2020-03-07:
I noticed the same issue with Sleep and tried to install/build it with prior versions. It is definitely the problem. I submitted a bug last night but removing the dependency would be the best thing to do.

I assume it is for timing.....how about a callback instead?

ksteinberg, 2020-03-07:
Or use another npm like delay or sleep-sync

jcane86, 2020-03-07:
yes, that was my initial thought, the async methods with callbacks are
already in the library, but to remove the sync stuff I'd have to deprecate
half the api, and I'm not sure if people are already relying on that.

I'll have look at sleep-sync and see if that helps.

just to clarify, you're saying that installing motor-hat in node 12 also
failed?

ksteinberg, 2020-03-07:
I did not try Node 12 as I am afraid it will impact my other npm modules....I kind of need to move forward with the existing node version.

I think using sleep-sync might be best...

ksteinberg, 2020-03-07:
so I tested sleep....it's working.  The issue is in your code somewhere....

I created a completely new project, new dir...installed the min....my sleep test script works fine even though the sleep npm install complained. 

Sorry, the issue is still in motor-hat :(

***********
code
**********
var sleep = require('sleep');
require('log-timestamp');

console.log('Before log-timestamp');
sleep.sleep(10);
console.log('After log-timestamp');

*****
output
*****

pi@raspberrypi:~/Public/servo $ sudo node sleep-test.js
[2020-03-07T12:04:27.550Z] Before log-timestamp
[2020-03-07T12:04:37.574Z] After log-timestamp

********
npm list
*******

├─┬ log-timestamp@0.3.0
│ └── log-prefix@0.1.1
├─┬ motor-hat@1.0.0-semanticrelease (github:jcane86/motor-hat#f64a863e6ad02a19f731df489c3b7bb15f54ba8e)
│ ├── async@3.2.0
│ ├─┬ debug@4.1.1
│ │ └── ms@2.1.2
│ ├─┬ i2c-bus@5.1.0
│ │ ├─┬ bindings@1.5.0
│ │ │ └── file-uri-to-path@1.0.0
│ │ └── nan@2.14.0
│ ├─┬ parambulator@1.5.2
│ │ ├─┬ gex@0.2.2
│ │ │ └── lodash@3.10.0 extraneous
│ │ ├── jsonic@0.2.2
│ │ └── lodash@4.5.0
│ ├── raspi-ver@1.1.1
│ └─┬ sleep@6.1.0
│   └── nan@2.14.0 deduped
└── stable@0.1.8

jcane86, 2020-03-07:
oh, wait, I've just gone through your last output again, and I noticed the warnings for sleep didn't actually stop the install, they were just warnings. it looks like that was a "successful" install.

did you try using the library at that point? what was the error?

(I'm referring to this comment, https://github.com/jcane86/motor-hat/issues/120#issuecomment-595927619 , when you first installed from the branch.

ksteinberg, 2020-03-07:
I did try it....it failed....

perhaps the bindings, as shown in my log 2 days ago?

ksteinberg, 2020-03-07:
here is the latest...code on top, output below

*******************
pi@raspberrypi:~/Public/servo $ more servo-test.js

var motorHat = require('motor-hat')({steppers: [{ W1: 'M1', W2: 'M2' }]}).init();
motorHat.steppers[0].setSpeed({pps:100});
motorHat.steppers[0].step('back', 2048, (err, result) => {
  if (err) return console.log('Oh no, there was an error', err);
  console.log(`Did ${result.steps} steps ${result.dir} in ${result.duration/1000} seconds. I had to retry ${result.retried} steps because you set me up qu
icker than your poor board can handle.`);
});
pi@raspberrypi:~/Public/servo $ sudo node servo-test.js
internal/modules/cjs/loader.js:323
      throw err;
      ^

Error: Cannot find module '/home/pi/Public/servo/node_modules/motor-hat/dist/index.js'. Please verify that the package.json has a valid "main" entry
    at tryPackage (internal/modules/cjs/loader.js:315:19)
    at Function.Module._findPath (internal/modules/cjs/loader.js:703:18)
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:967:27)
    at Function.Module._load (internal/modules/cjs/loader.js:862:27)
    at Module.require (internal/modules/cjs/loader.js:1040:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/home/pi/Public/servo/servo-test.js:2:16)
    at Module._compile (internal/modules/cjs/loader.js:1151:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1171:10)
    at Module.load (internal/modules/cjs/loader.js:1000:32) {
  code: 'MODULE_NOT_FOUND',
  path: '/home/pi/Public/servo/node_modules/motor-hat/package.json',
  requestPath: 'motor-hat'
}
pi@raspberrypi:~/Public/servo $

jcane86, 2020-03-07:
ah, sorry, that might be because you're installing from the branch, and there's an extra build step that gets done in CI when I release a version.

sorry for all this hassle, could you try cd-ing into node_modules/motor-hat and running npm run build?

that should compile all the code into the dist folder, and get rid of that last error

jcane86, 2020-03-07:
you might have to do an npm install there first, to get the devDependencies

jcane86, 2020-03-27:
@ksteinberg , I've just published motor-hat 2.0.10 on npm, which should hopefully solve this problem.. please give it a try and let me know if you're still having any issues.

Cheers

Pepe

closed 2020-09-03 by jcane86