jcane86/motor-hat #122: Installation failing on Raspberry Pi 4, node v12.16.1, npm 6.13.4

nrbrt, 2020-03-18:
I have seen that there is another ticket open for node 13.9.0, but I thought it was worth opening a new ticket for the stable version of node. It starts with the i2c-bus version. That does not work. I have tried forking ang changing the version to 5.1.0, but that opens a whole new can of worms. Is there any way to get this working? What version of node does it work on?

nrbrt, 2020-03-19:
I manually copied the lib directory to the node_modules/motor-hat directory, together with the test directory and now npm run build completes successfully.
I tried running your the basic usage script and it showed me this:
```
/home/pi/node-projects/motor/node_modules/motor-hat/node_modules/i2c-bus/i2c-bus.js:423
    i2c.writeByteSync(peripheralSync(this, addr), cmd, byte);
        ^

Error: , Remote I/O error
    at Bus.writeByteSync (/home/pi/node-projects/motor/node_modules/motor-hat/node_modules/i2c-bus/i2c-bus.js:423:9)
    at setAllPWMSync (/home/pi/node-projects/motor/node_modules/motor-hat/dist/pwm.js:203:9)
    at Object.init (/home/pi/node-projects/motor/node_modules/motor-hat/dist/pwm.js:244:7)
    at Object.init (/home/pi/node-projects/motor/node_modules/motor-hat/dist/index.js:219:11)
    at Object.<anonymous> (/home/pi/node-projects/motor/test.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:1158:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)
    at Module.load (internal/modules/cjs/loader.js:1002:32)
    at Function.Module._load (internal/modules/cjs/loader.js:901:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12) {
  errno: 121,
  code: '',
  syscall: 'writeByteSync'
}
```
That might be ok, since I do not actually have a motor hat attached to my pi. As soon as I get this going, I will try and create a node-red implementation using your module.

jcane86, 2020-03-19:
Hi,

Yeah, that last error looks like an actual i2c communication error, which
would make sense if the board is not actually there to talk to.

I'll try to get a new version out this weekend with the latest i2c-bus
dependency, that should hopefully fix installation problems...

I'll report back here once I'm able to release the new version

nrbrt, 2020-03-19:
awesome, thanks!

jcane86, 2020-03-26:
hi @nrbrt , just thought I'd give a quick update.

Sorry I haven't been able to cut the new release yet, but I'm working on it. Should be out by this weekend for sure.

Cheers!

jcane86, 2020-03-27:
@nrbrt , I've just published v2.0.10 of motor-hat, which should solve your initial problem. If you install directly from npm now it should work for you (as long as you have the motor hat connected).

I'll close this issue for now, feel free to reopen if you're still having problems with this.

Cheers

Pepe

closed 2020-03-27 by jcane86