jcane86/motor-hat #123: Id not recognized

nrbrt, 2020-04-03:
Hi, thanks for updating your module! It installs perfectly, but when I started using it, it told me to get back in touch with you :-)

```
Error: We dont recognize your revision (Revision	: c03112,c03112). Sorry! get in touch and let me know so I'll add it
```
Let me know if I can do anything to help.

jcane86, 2020-04-03:
Hi @nrbrt looks like you got a new raspberry that my other lib didn't cover yet. If you uninstall motor-hat and reinstall it again, it should hopefully work now.

Sorry for the hassle, and thanks for reporting!

nrbrt, 2020-04-03:
Not a problem! The support is awesome ;-)

nrbrt, 2020-04-06:
ok, that works now. Thanks a lot! However, I have run into the next hurdle:
```
/home/pi/node-projects/motor/node_modules/i2c-bus/i2c-bus.js:423
    i2c.writeByteSync(peripheralSync(this, addr), cmd, byte);
        ^

Error: , Remote I/O error
    at Bus.writeByteSync (/home/pi/node-projects/motor/node_modules/i2c-bus/i2c-bus.js:423:9)
    at setAllPWMSync (/home/pi/node-projects/motor/node_modules/motor-hat/dist/pwm.js:203:9)
    at Object.init (/home/pi/node-projects/motor/node_modules/motor-hat/dist/pwm.js:244:7)
    at Object.init (/home/pi/node-projects/motor/node_modules/motor-hat/dist/index.js:219:11)
    at Object.<anonymous> (/home/pi/node-projects/motor/index.js:1:75)
    at Module._compile (internal/modules/cjs/loader.js:1158:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1178:10)
    at Module.load (internal/modules/cjs/loader.js:1002:32)
    at Function.Module._load (internal/modules/cjs/loader.js:901:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12) {
  errno: 121,
  code: '',
  syscall: 'writeByteSync'
}
```
That happens when I try to run your example code:

```
var motorHat = require('motor-hat')({steppers: [{ W1: 'M1', W2: 'M2' }]}).init();
motorHat.steppers[0].setSpeed({pps:100});
motorHat.steppers[0].step('back', 2048, (err, result) => {
  if (err) return console.log('Oh no, there was an error', err);
  console.log(`Did ${result.steps} steps ${result.dir} in ${result.duration/1000} seconds. I had to retry ${result.retried} steps because you set me up quicker than your poor board can handle.`); 
});
```
Any ideas?

jcane86, 2020-04-06:
hi @nrbt, yeah, that looks like a common issue with the i2c address being different from the default.

see here #18.

I suggest trying to set the i2c to 0x60,that seems to be a common one

jcane86, 2020-04-07:
@nrbrt basically add the following to the example:
```diff
var motorHat = require('motor-hat')({
+  address: 0x60,
  steppers: [
    ['M1', 'M2'],
    ['M3', 'M4']
  ]
});
```

jcane86, 2020-04-07:
BTW, PRs to improve the readme to make this clearer are welcome

nrbrt, 2020-04-07:
Wow, I completely missed that one. I think the address selection should be taken out of the "advanced usage" section. My idea was to learn how to walk before doing any running, so I ignored the advanced usage section for the moment and missed that crucial part. I will have a look and see how I can help and contribute to your project. Thanks again!

closed 2020-04-07 by nrbrt